TheSystemDevelopmentKit/spice #4: Get rid of VerilogA

sporrasm, 2021-01-20:
The current way of reading the output results is directly from simulator using VerilogA (applies to Spectre). We should probably get rid of this and instead implement reading the results from the output database using Python instead.

Making the file reading general w.r.t. analysis type might be difficult for Spectre, as the output format can vary. To the best of my knowledge, the output format can be somewhat influenced with the simulator options.

mkosunen, 2021-04-10:
Ping @sporrasm @ojarvin @kspoof 
Solution
```
Option1 options printstep=yes
...
TRAN_analysis tran pstep=1e-12 stop=1.024e-05 RAN_analysis tran pstep=1e-12 stop=1.024e-05 
 
simulator lang=spice
.option ingold=2    
.print tran v(outtiputti)
simulator lang=spectre
...
```

We discussed about this, and after considerations I think this methodology should be the default, as it is
1) Supported in all of the eldo-like simulatiors quite similarly. 
2) It supports all analysis types similarily.

Consequently, current verilogA sampling should be renamed to something else, OR we should keep the current name, figure a descriptive name for this, and make it a default. That would preserve backwards compatibility. What is the name for the current verilogA/eldo sampling format. Is it 'sample' or 'event'?

mkosunen, 2021-04-10:
Name seems to be be 'event'.

ojarvin, 2021-04-10:
It seems safe yes.

However, the 'time' type outputs process the information from an 'event' type output. So too coarse time steps in the continuous signal type will result in bad time-domain info extracted from them.

Can this print statement write all calculated points to file? In other words, does the print time step **have** to be defined?

If so, I think the 'event' type verilogA writer can be directly replaced. If not, some tests should be done whether there is a good default value that doesn't result in insane file sizes or poor results.

Also, the currents should be writable also, since the transient power extraction uses that. I assume this print can print anything, so it shouldn't be a problem.

mkosunen, 2021-04-10:
ÄÄÄH!  Spectre doesn't seem to allow to define the output file names.

PR #32 (2021-10-01): Getting rid of VerilogA
Spectre runs VerilogA-free in this branch now.

Tested for spice_iofile types
- event (was already done)
- sample, for both digital (ioformat='bin' or 'dec') and voltage (ioformat='volt')
- time

Each sample (each bit in a bus) or time signal gets saved as an event-type print from spectre, and all further parsing is done in Python. This should make it quite trivial to extend to Eldo and ngspice as well (as long as they support event IOs without veriloga). Downside is again larger output filesize.

I also deprecated spice_iofile type 'vsample' because in my opinion it is [... cut]
closed 2021-10-01 by ojarvin