TheSystemDevelopmentKit/spice #23: Interactive spice source locations

mkosunen, 2021-04-10:
@ojarvin 
s it intentional that interactive simulations have different tb and soure locations than other ismulations?
```
@property
    def spicesubcktsrc(self):
        """String

        Path to the parsed subcircuit file. ('./spice/subckt_entityname.scs').
        This shouldn't be set manually.
        """
        if not hasattr(self, '_spicesubcktsrc'):
            if self.interactive_spice:
                self._spicesubcktsrc=self.spicesrcpath + '/subckt_' + self.name + self.syntaxdict["cmdfile_ext"]
            else:
                self._spicesubcktsrc=self.spicesimpath + '/subckt_' + self.name + self.syntaxdict["cmdfile_ext"]
        return self._spicesubcktsrc

```

ojarvin, 2021-04-10:
It is intentional, but probably could be done in some other way.

The reason for this is having the simulation database open in ezwave (or viva) **during** a long interactive simulation is nice for monitoring the results in real time. In this case, the database is saved under `./spice/` I believe. The convenience here is that restarting the simulation will just overwrite the database, and ezwave can be kept open and it will refresh the waveforms.

For non-interactive simulations, the databases are separated into the same directories as the other input and output files (`./Simulations/spicesim/<timestamp>/` or something like that). This is because of possible parallel simulations and avoiding overlap. Technically the database can go here even for interactive simulations, but then restarting the viewing software and navigating to the new file location every time would be very tedious.

I was thinking about having the spice module automatically launch ezwave or viva as the database is created (during the simulation). However, I wouldn't want this to happen every time either because this would turn into a game of whack-a-mole in terms of closing all kinds of viva/ezwave windows popping up for every simulation.

mkosunen, 2021-04-10:
OK. Got it.

closed 2021-04-10 by mkosunen