TheSystemDevelopmentKit/spice #47: Unifor sampling in spectre

mkosunen, 2022-10-31:
It sould be tested afain with a very simple structure that this strobeoutput=strobeonly has no effect. If this really is the case, the problem could be reported to Cadence.
If it is not effective, and there is clutter in the result, the solution is to write a clutter removal function in python. I understood, that the time values are still sorted and that the probed time instances really are in the file, so the fastest possible method to remove clutter is OK.

mkosunen, 2022-10-31:
Ping @sporrasm @vlahtin

mkosunen, 2022-12-30:
@vlahtin can you verify this? If this is the case, cloes this.

vlahtin, 2022-12-30:
Yep, no effective solution to solve uniform sampling was found from simulator side, and this is done with an implemented clutter removal function.

vlahtin, 2022-12-31:
An issue I found with this method is that at least when doing very large number of decimal (very small floating point number strobeperiod), the simulator does not strobe the values at the correct positions. At least with a NR siggen setup with baseband sampling frequency of 245760000Hz, and strobeperiod of 9.934107462565105e-13, it seems like the error is growing to be larger and larger towards the final values. 

For instance in this simulation, the final value in this simulation should have been 1.8762523929278056e-05, but the closest value to this in the print file was 1.87625239263637e-05. Will have to investigate this further, and find out how significant this issue is. Currently my hypothesis is that this may be the case of EVM calculation degradation.

The result does not seem to affect the spectrum, however this could be the reason for the EVM phase error, if the delay is significant enough. I will check this soon as I get simulations run to this point again in Python 

ping @mkosunen @sporrasm

vlahtin, 2022-12-31:
The filter however, in this case seems to be working correctly, and selecting the closest value. The issue is that the correct value does not exist in the print file. I will rerun the simulation and add more digits to the simulation to see if this issue is solved

mkosunen, 2023-01-01:
Observing this accumulating timing error is a good catch. It seems to me that for this simulator we need the postprocessing anyway, and the simplest way would be linear interpolation (used in ngspice for example).

However, this does not obsolete the fractional rate interpolation as we need it  for the receivers and transmitters anyway.

We need to cath up sometime soon, as where this is implemented is related to ongoing refactoring of spice.

vlahtin, 2023-01-04:
Upon further review, this does not contribute to any significant errors on the simulation. On further thoughts, if there was a change happening at these non-included points the simulator would probably calculate the values at these points.

The EVM issue was caused by the testbench generation issue. The timeperiod used to generate the IOs have effective accuracy of about 6 floating point digits. I think this mentioned comment of the mismatching strobepoints is something to keep in mind, but it does not contribute any degradation in EVM or spectrums at least in my case. 

Unless @sporrasm @mkosunen you have anything to comment on this, this issue can be  closed in my opinion

vlahtin, 2023-01-05:
The aforementioned timeperiod rounding is now fixed in #57. 

As this seems to work as it should and the strobeperiod error does not contribute to errors in spectrum or EVM I will close this issue.

closed 2023-01-05 by vlahtin