TheSystemDevelopmentKit/spice #51: Set strobed to True by default by v2.0

sporrasm, 2022-11-18:


mkosunen, 2022-12-30:
@sporrasm @vlahtin :  While working on #52 , I noticed that it remains a bit unclear how the outputs for analog signals should be preseneted. Then I found this issue opened by @sporrasm and rmembered that we've been discussing this, so I write here some plan that can be reformulated in the future.

My thoughts:
1. Inputs: AFAIK the analog simulator piecewise linear sources need time voltage pairs, which is 'event' type input regardless whether the time interval is uniform or not. However, most often the input is uniformly sampled as uniformly sampled signal is convenient to handle, and is the elementary way of presenting signals in signal processing. Non-uniform sampling is a bit more advanced concept. Therefore I suggest that we default the inputs of analog simulators to 'event' type. Sampling does not need to be uniform, but often it is.

2. Outputs: Because _it is convenient_ to have uniformly sampled signals at the output too, I suggest that we _aim_ for that in a manner that the uniformly sampled signal is provided already by the simulator, but as an 'event' type IO. There is methodology in Eldo and Ngspice (tstep/coustep parameter in .tran ) to achieve this, and it is also documented to consume less memory for ngspice. It also provides smaller IO-files. For Spectre it has turned out to be trickier, and mandates postprocessing. At least in Ngspice the method used to provide uniform samples is linear interpolation. Nothing more complex. Alternatively we need to have postprocessing (at least linar interpolation) for all simulators.

To conclude the suggestion: 

1. The type of the input and the output for analog simulator analog signal is 'event'
2. The uniform sampling for output is implemented in the testbench. For Spectre this also requires postprocessing which seems to be unique to Spectre (picking the correct samples fro cluttered outputs, not interpolation).
3. Uniform sampling mode can be is disabled  by setting 'native_sampling' flag to true, which prevents/removes the sampling interval options from the .tran . it is then up to designer to handle whatever comes out from the simulator, and that is NOT implemented as a part of spice module, as it is just a general method to handle nonuniformly sampled signals.

Let me know what you think.

mkosunen, 2022-12-30:
Related !19

closed 2023-03-25 by mkosunen