TheSystemDevelopmentKit/spice #64: Spectre AC simulation output fail read fails if the results are in different arrays

vlahtin, 2023-05-03:
If the designer is using multiple output iofiles (e.g. over 3), the real and imaginary part of the output may end up on different arrays, which means that the real and imaginary data are not combined correctly (lines 315-316 in spice/spice_iofile.py). 

The read works fine, if the number of output iofiles is 2 or 1, as the output print file does not get split into multiple rows. 

@sporrasm @mkosunen

sporrasm, 2023-05-04:
As discussed, we could transferring the spectre file reading methodology to something more reasonable than print file. My suggestion is to use Cadence in-built command line tool: psf. There is also a Python library for this, if we want to avoid coding in the interface: https://pypi.org/project/libpsf/.

If we want to go this route, first step would be try how slow it is in comparison to current file reading and see if it improves anything.

sporrasm, 2023-06-13:
I believe this if fixed by commit cc7d495 of v1.10_RC (tested). However, see #72. This is not a deal breaker for AC sims, but something to take a look at.

closed 2023-06-13 by sporrasm